LK8000/LK8000 #1443: CUP File parser quits at lines >=2048 (LK8000 beta)

kredo01, 2020-04-30:
The waypoint parser stops if in the CUP file lines >= 2046 characters exists.
The prevoius waypoints are shown up to the affected line, but no information regarding troubles.
I luckly was missing an aerodrome, otherwise I wouldn't recognize the problem.

I ran into that problem using the current Austrian cup file from OFM.
At aerodrome info (LOLW) caused that problem.

Would be helpful to throw some errors in that case.

The problem exists only in LK8000 beta. Previous verisions obviously work fine with lines >=2048.

brunotl, 2020-04-30:
please add a turmpoint file to this tickets to replicate the issues.

thanks

AlphaLima, 2020-04-30:
https://github.com/AlphaLima/LK8000_Data/blob/master/_Waypoints/OFM_CUP/austria_ofm.cup
Line 571, 
the buffer for readline is limited to
READLINE_LENGTH (1024)
the problem is, that reading the file is abort at ths point. All following information is lost.
increasing the buffer is not the solution,
we need a error handler for
zzip_stream
instances

brunotl, 2020-04-30:
for Waypoint, line size limit is 2048 : 
`static TCHAR nTemp2String[READLINE_LENGTH*2];`

a workaround is to ignore the end of  line instead of return error...

AlphaLima, 2020-04-30:
and how do we determin the end of the file, 
as a effect we ignore this waypoint with the long comment, without notice to the user.
same problem exist for CUP task loading

brunotl, 2020-04-30:
we don't ignore waypoint with long comment we only truncate the comment.

AlphaLima, 2020-04-30:
really?
if yes, it is a very good solution.

closed 2020-04-30 by brunotl

kredo01, 2020-05-02:
> we don't ignore waypoint with long comment we only truncate the comment.

Hi Bruno!

But is seems, that also the following waypoints are truncated!?!

Using the current Austrian (LO) cup file from OFM, you will find at line 2754 "Wels LOLW" with a line lenght of 2312 chars.
[ofm.txt](https://github.com/LK8000/LK8000/files/4567309/ofm.txt)

This causes, that all the following waypoints are not shown. No notice for the user.

The problem apperas at Bete. Not reprodurceable at previous versions.

best regards and happy landings!

AlphaLima, 2020-05-02:
This problem is already fixed for the next beta version.