LK8000/LK8000 #1452: Maximum number of Polar files? - and _Polar file

JBlyth, 2020-05-17:
With LK8000 PC 7.1n version I seem to have found there is a maximum number of Polar files allowed. I think it is somewhere between 280-300 .plr files. If there are more than that in the folder, the last ones are not shown in the LK8000 Aircraft Polar listing. The Android version seems to accept more polar files.

There is also a blank file _POLARS in the _Polars sub-directory. If I remove this, I get an error on starting LK, but not with the Android version.....

Why do I have so many polars? It is because I have the polar files from LK and "XCSoar" format ones from Condor 2, but they are all different names, so there is not a clash.

The work around is easy - just remove some of the Polar files, but I thought I should flag the apparent limitation.

Cheers
John in Sunny Carnarvon - Western Australia

brunotl, 2020-05-18:
limit is 300, same for all files chooser and all platform. no reason to change that.

XCSoar and LK8000 default polar are almost the same ( in most case only file name is different ) so it's useless to have both ...

if some polar is missing in our default polar, you can provide files and ask to add it.

closed 2020-05-18 by brunotl

JBlyth, 2020-05-18:
Thanks Bruno,

Good information to know.

Cheers
John