LK8000/LK8000 #1454: New feature: waypoint display GR + arrival altitude

bobmorane06, 2020-05-29:
Hello community, although last time that I coded was in 1996 (!), I decided to try to clone LK8000 and add a feature that I was missing in LK8000. 

I added a new Waypoint display mode with glide ratio required *and* terrain glide with the format “WaypointName : GR/AltArrival”. 

The reason behind the feature is that I find it safer to make my in-flight navigation decisions based on GR (eg I decide not to go further than GR 20 from the closest landable on a glider with theoretical L/D of 40), but the arrival altitude adds confidence. Today it’s one or the other so I added an entry to "Landable value" in menu "11 Appearance".
 
[image]

I still have a tweak to add to the code, then I’ll do my best to create a Pull Request this weekend (I’m still struggling with git), assuming you think it makes sense to add my change.

brunotl, 2020-05-30:
do a draft pullrequest with your change, like that we can do a code review ...

bobmorane06, 2020-05-30:
Done, it was tedious and I hope I did it right :-) 

I wasn't too sure of coding style and the use of tabs and/or spaces, I tried to mimic the existing code

PR #1455 (2020-06-05): Add new Landables value: GR + arrival altitude
Adds an additional mode for the display of “Landable value”,
comprising both geometrical Glide Ratio required and arrival altitude
above safety with format “WaypointName : GR/AltArrival”.

Motivation: I find it safer to make my in-flight navigation decisions
based on GR required (eg I decide not to go further than GR 20 from
the closest landable on a glider with theoretical L/D of 40), but the
arrival altitude adds confidence. Today it’s one or the other so I
added an entry to have both displayed simultaneously.

- Add avGR_Altitude to avAltitude/ [... cut]
commit 8ac7567f (2020-06-05): Add new Landables value: GR + arrival altitude

Adds an additional mode for the display of “Landable value”,
comprising both geometrical Glide Ratio required and arrival altitude
above safety with format “WaypointName : GR/AltArrival”.

Motivation: I find it safer to make my in-flight navigation decisions
based on GR required (eg I decide not to go further than GR 20 from
the closest landable on a glider with theoretical L/D of 40), but the
arrival altitude adds confidence. Today it’s one or the other so I
added an entry to have both displayed simultaneously.

- Add avGR_Altitude to avAltitude [... cut]
closed 2020-06-05 by brunotl