LK8000/LK8000 #1457: [Proposal] Update Wiki to add WSL2 build environment

bobmorane06, 2020-06-02:
HI folks, I just installed WSL2 / Windows Subsystem for Linux today on my Windows machine and successfully built for Linux and PC. It is very straightforward and once setup, faster than booting a VM. I suggest to edit the LK8000/wiki/Windows page by adding a WSL2 section.

It would be as on https://github.com/bobmorane06/LK8000/wiki  (only added the section "Windows Subsystem for Linux / WSL2")

Should I edit the LK8000 page directly? 

FYI I setup an Ubuntu 20.0.4 LTS and had to install mingw-w64 as mingw32 wouldn't install.

I noticed when building that it only used circa 50% of my CPU cores, I haven't looked yet at how to increase that (building on my Ubuntu Virtual Box is faster as I assigned more CPU cores)

bobmorane06, 2020-06-02:
Ignore my comment about number of CPU cores. Running make with -j16 uses 100% of my Windows CPU usage

brunotl, 2020-06-03:
i think you can edit LK8000 page directly.

 some comments : 
- cygwin and mingw section are outdated.
- WSL2 (or Docker container) could be a nice solution but how to run binary inside debugger ?
     after some time spend on, i'm back to cygwin for that...
- instead of Ubuntu, if you use debian 10, i think you can also use our prebuild Kobo  & WinCE toolchain availabale here : http://lk8000.it/toolchain/ ( carefull about kobo if build is not made on linux filesystem you can't run it on device ... )

brunotl, 2022-01-27:
wiki updated

closed 2022-01-27 by brunotl