LK8000/LK8000 #1543: OUDIE2 LITE BLUETOOTH

K-Factor-1961, 2021-05-30:
I discovered that with last version of OUDIE 2 Lite, if you want to install LK8000 it is impossible to activate the internal bluetooth device to connect the unit to an external device. 
It seems that it is possible to activate bluetooth (on com5) only by seeyou mobile application and when you go out the application, the bluetooth device is swutched off.
Which may be the solution?

brunotl, 2021-05-30:
BT is automaticly activated when you pair the device in LK8000 device config.

K-Factor-1961, 2021-05-31:
If you try to install on a new Oudie2 the LK8000 software you have not options in device config to set bluetooth.

brunotl, 2021-05-31:
What LK8000 version ?

brunotl, 2021-05-31:
in device config you have "Bluetooth" button....

[image]

K-Factor-1961, 2021-06-01:
Hi Bruno.
I installed last LK8000 version 7.2.1 Yes in the previous version of OUDIE2 FW this is right. With the new version of OUDIE2 FW this option is not avaiable.

K-Factor-1961, 2021-06-01:
You can try upgrading your OUDIE2 tyo last FW version

brunotl, 2021-06-01:
if you don't see "Bluetooth" button "BTDrv.dll" or "BTDrt.dll" is missing on the device ( removed by upgrade ? ) ...

K-Factor-1961, 2021-06-02:
I cannot understan. If you have OUDIE" in your hand try to upgrade the fw and check it. 
Why seeyou mobile works correctly with blue tooth?
Or you can isert the "BTDrv.dll" or "BTDrt.dll" by LX8000 installation?

brunotl, 2021-06-02:
oudie & seeyou are made by naviter, so it's not a surprise if seeyou mobile can use oudie bluetooth....
the dll are parts of WinCE OS we can't insert it. 

so it seem's new oudie don't use the same way as previous to manage bluetooth, without information from naviter we can't know how to do it...

brunotl, 2026-02-27:
WinCE support is end since 7.5.1