LK8000/LK8000 #1544: Copyright violation by using the "okio" library

MaxKellermann, 2021-06-29:
LK8000 is licensed under GPLv2, but LK8000 for Android is distributed with the "okio" library, which uses the Apache2 license. This license is not compatible with GPLv2.
See https://www.apache.org/licenses/GPL-compatibility.html
https://www.gnu.org/licenses/license-list.en.html#apache2

okio license: https://github.com/square/okio/blob/master/LICENSE.txt
LK8000 license (the "or later" clause present in XCSoar has been removed): https://github.com/LK8000/LK8000/blob/d9b74b03f596368d34a0bcfbaab2e2c6206a1516/android/src/org/LK8000/UsbSerialPort.java#L3

Therefore, distributing this combined work of GPLv2 and Apache2 code is a copyright violation.

This can be fixed by relicensing LK8000 as GPLv3.

MaxKellermann, 2021-06-30:
There's another copyright violation due to Apache 2 license: the `com.google.zxing` library, used by `class QRCodeImageAnalyzer`, is Apache2-licensed as well.

brunotl, 2021-07-27:
closed by commit 6ddd613

closed 2021-07-27 by brunotl