LK8000/LK8000 #1547: Round airspaces doesn't work in the v7.2.1

kubasto, 2021-07-10:
I'm not sure if it's just a matter of new version, but I've made a series of test and I can't see other reason for the problem.
Airspaces like this one below doesn't work in in v7.2.1, but works fine in v7.1m:

```
AC Q
AN Mala_Czantoria
AL GND
AH UNLIMITED
V X=49:41:34 N 18:46.4342 E
DC 0.21598
```
My other airspaces work fine, however they have different structure usually:

```
AC G
AN TR 69E
AL GND
AH 24500F MSL
DP 53:28:56 N 016:17:23 E
DP 53:17:51 N 016:23:06 E
DP 53:13:54 N 016:36:44 E
DP 53:23:03 N 016:34:00 E
DP 53:28:56 N 016:17:23 E
```
I've checked:

- config > 2 > airspace display: All on
- config > 2 > filter (all are set to display)

It's like the first airspace doesn't exist.

brunotl, 2021-07-11:
if the last airspace of file is circle it is not loaded correctly, already fixed by commit 0f6eb066

closed 2021-07-11 by brunotl

kubasto, 2021-07-12:
Thanks. I've checked the 7.3.7, and although the airspace can now be found by [what's here] tool in the pan mode, the circles are still not drawn on the map. Am I missing something?

brunotl, 2021-07-13:
https://www.postfrontal.com/forum/topic.asp?TOPIC_ID=9247