LK8000/LK8000 #1549: Add Borgelt B800 device driver

brunotl, 2021-07-29:
https://www.postfrontal.com/forum/topic.asp?TOPIC_ID=9254

SorenBraes, 2021-08-12:
Hi all,
after several conversations with Borgelt Instruments, we now have the outbound protocol used. I have attached this as a PDF document. I also attached a log file from the serial output of an existing B800 vario, where I varied the NcCready, bugs and ballast. The exact settings can be found in the comment column of the Excel sheet,
Hope this helps in creating a full B50/B800 interface. Don't hesitate to contact me, if I can be of any help testing or giving more info.
Best regards
Soren Braes[
[Description of the serial data stream coming out of the Borgelt B50.pdf](https://github.com/LK8000/LK8000/files/6977681/Description.of.the.serial.data.stream.coming.out.of.the.Borgelt.B50.pdf)
[B800 Output Protocol - LK8000.xlsx](https://github.com/LK8000/LK8000/files/6977683/B800.Output.Protocol.-.LK8000.xlsx)
](url)

SorenBraes, 2021-08-12:
Sorry, forgot to write, that the inbound (PDA/PNA -> Vario) protocol is Cambridge 302.