LK8000/LK8000 #1550: Change maximum waypoint radius admited on LK8000 from 100Km to as is possile

Mrapari, 2021-08-02:
During last Italian campionship I've seen that the maximum waypoint radius admited on LK8000 is 100Km, and for this reason I couldn't use LK because some waypoints radius on task competition was upper 100Km. I think it is easy to solve and I'll be very gratefull if it will done. Thanks so much

AlphaLima, 2021-08-02:
what version are you you using?
with the latest beta 7.3.8, I don't have any radius limitation, not in normal sectors nor cylindwers.
Testet for normal and AAT Tasks.

[image]

Mrapari, 2021-08-02:
Ho, the problem is shown only for the start point. The other seems to be good

AlphaLima, 2021-08-02:
A start radius > 100km ???
(Does not make sense to me)

Mrapari, 2021-08-02:
I don't know the sense, but it was in the last Italian champions. And for that reason I can't use LK8000 for that tasks

brunotl, 2021-08-02:
> I don't know the sense, but it was in the last Italian champions. And for that reason I can't use LK8000 for that tasks

can you share the task details ?

Mrapari, 2021-08-02:
Here one of the task planned on xctrack..

brunotl, 2021-08-02:
> Here one of the task planned on xctrack..

where is the task ?

Mrapari, 2021-08-02:
File attached in the previous mail

brunotl, 2021-08-02:
no attachement in github issues comments ...

Mrapari, 2021-08-02:
Sorry I've attached in the mail. but now added also on this post
[image]
[image]

Mrapari, 2021-08-02:
Also a waypoint file if it is necessary..
[Trofeo 100 Laghi.zip](https://github.com/LK8000/LK8000/files/6917817/Trofeo.100.Laghi.zip)

brunotl, 2021-08-05:
tips : if task is loaded from xctrack tsk file or QRCode, radius are right still you don't edit the turn point.

brunotl, 2021-08-05:
Closed by commit 6a9cb9b56003d38835e600c08207d42096b6d858

closed 2021-08-05 by brunotl

Mrapari, 2021-08-05:
Yes, but it isn't the better way to do competition. Frequently the tasks can be varied during and after the breefing...