LK8000/LK8000 #1556: Characters in Task Waypoints above UTF-8 127?

JBlyth, 2021-09-26:
I came across a strange problem when loading a task translated for Condor 2 via CoTaCo today. I reverted back to 7.2.1 and the same thing happened.

One of the Waypoint names had an **ó** in "Lago Espol**ó**n", and this seemed to stop the task file from loading that waypoint and those following.

Changing the "**ó**" to "**o**" resolves the issue. I expect that any other character with an accent may be a problem as well.

Is this a known limitation in Waypoint names? I presume it is by design as the LK keyboard itself doesn't have those characters. Does anyone have a smart way of trapping invalid characters in Waypoint names that I can pass on to Marc who developed CoTaCo?

Many thanks
John in Sunny Carnarvon - Western Australia

brunotl, 2021-09-26:
Join task file to the ticket...

JBlyth, 2021-09-26:
Sorry, I should have thought of that... Here you are
[Default.txt](https://github.com/LK8000/LK8000/files/7231172/Default.txt)
.

brunotl, 2021-09-27:
you file is not valid utf8 xml file, so it's CoTaCo bug ....
[image]

closed 2021-09-27 by brunotl

Bre901, 2021-09-27:
Maybe it's a CoTaCo bug (the file is indeed encoded in Windows-1252, not in UTF8), but...

How does one specify another encoding in the file header ? 
I have tried "encoding=ANSI" (which is the term generally used for 1252), but to no avail.
Maybe if the lkt format was thoroughly documented (try googling lkt "file format") these issues could be solved faster ...

brunotl, 2021-09-27:
documentation is available : 
- lkt file in our wiki : https://github.com/LK8000/LK8000/wiki/TaskFile
- xml format on w3c website : https://www.w3.org/TR/xml11/

i don't have time to work on workaround for bugge software so : 
- ask to CoTaCo to fix the bug
- in mean time, fix the file encoding yourself using your favorite text editor...

brunotl, 2021-09-27:
@Bre901 if i don't make mistake CoTaCo is your software...
2 tips to help you to fix the problem : 
  - LK8000 use rapidxml library to load & save task
  - only utf-8 encoded xml is supported
  - using ANSI code page is not a solution if your software could be used by people outside of western europe...

Bre901, 2021-09-27:
Talking about bugged software : why doesn't LK8000 accept windows-1252 characters in XML files when the encoding is correctly specified : 
<?xml version="1.0" encoding="ISO-8859-1"?> or <?xml version="1.0" encoding="windows-1252"?>
https://www.alchemysoftware.com/livedocs/cat9/xml/how_xml_deals_with_different_character_sets.htm
[ANSI1252.txt](https://github.com/LK8000/LK8000/files/7237488/ANSI1252.txt)

Bre901, 2021-09-27:
Our messages crossed each other

"only utf-8 encoded xml is supported" glad to learn that, where was I supposed to find that information ? it's not in the wiki. BTW why did you change that link ?

I'm using ANSI code page because Condor does so

brunotl, 2021-09-27:
> "only utf-8 encoded xml is supported" glad to learn that, where was I supposed to find that information ? it's not in the wiki.

right it's not as explicitly writed... ( but file  "should look like an exemple wish is utf8 encoded"...

we use utf-8 to allow using CJK, cyrylic or all other unicode characters not included in ansi code page ...
why to support ANSI and not the hunder's others ...  
lkt file format was made to be the internal LK8000 task file format, so the only need is to read and write the file using the same charset. 

> BTW why did you change the link to the wiki ?

when i did that ?

> I'm using ANSI code page because Condor does so

how they deal with turnpoint in containing cyrilic or CJK characters ?