LK8000/LK8000 #1559: Run LK8000 on Raspberry PI4 and Raspberry PI3

urb31075, 2022-01-07:
I compiled from source LK8000 on Paspberry Pi 4. I start it and everything works great.  I tried compiling to Raspberry Pi 3.  The program started without errors, but the buttons on the front panel are not displayed.
[image]

urb31075, 2022-01-07:
[RUNTIME.log](https://github.com/LK8000/LK8000/files/7830855/RUNTIME.log)

urb31075, 2022-01-07:
I compile on Raspberry Pi 3 with: 
make TARGET=LINUX DEBUG=y
make install DEBUG=y

On start i see:

LK8000-LINUX_debug: Common/Source/xcs/Screen/OpenGL/Canvas.cpp:717: const PixelSize Canvas::CalcTextSize(const TCHAR*) const: Assertion `font' failed.
Aborted

brunotl, 2022-01-07:
what is linux distribution ?

assert is about missing font, font needed is "ttf-dejavu", available in different place depending of linux distribution and not automatically installed by default.

list of known font are here : https://github.com/LK8000/LK8000/blob/983f2974ecd9120f4ba05f050b435e8e261cf36f/Common/Source/xcs/Screen/Custom/Files.cpp

urb31075, 2022-01-08:
1) ...what is linux distribution ? ...

I install OS in three steps
 - uploaded and written to the microsd image Raspberry Pi OS Lite (Release date: October 30th 2021)
   from https://www.raspberrypi.com/software/operating-systems/
 - configuring the system
 - Install the graphical shell with the command sudo apt install raspberrypi-ui-mods

This makes it possible to install a desktop raspbian without a lot of unnecessary programs.

2) Unfortunately, this installation method did not install the required fonts.

installing the fonts by the command:  sudo apt-get install fonts-dejavu solved my problems

I sincerely thank you for your help

closed 2022-01-08 by brunotl