LK8000/LK8000 #1573: Compile for Raspberry Pi OS (64-bit) error

urb31075, 2022-02-05:
I am trying to compile DL800 into Raspberry Pi OS (64-bit)
https://www.raspberrypi.com/software/operating-systems/

**efd:~$ uname -a
Linux efd 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux

efd:~$ gcc --version
gcc (Debian 10.2.1-6) 10.2.1 20210110**

I follow the commands:
**git clone https://github.com/LK8000/LK8000.git 
cd LK8000 
make TARGET=LINUX distrib**
I get an error:
Common/Source/MapDraw/DrawTerrain.cpp:879:5: note:   template argument deduction/substitution failed:
In file included from Common/Source/xcs/Math/Point2D.hpp:27,
                 from Common/Source/xcs/Screen/Custom/Point.hpp:27,
                 from Common/Source/xcs/Screen/OpenGL/Point.hpp:32,
                 from Common/Source/xcs/Screen/Point.hpp:28,
                 from Common/Header/linuxcompat/ScreenCoordinate.h:15,
                 from Common/Header/linuxcompat/types.h:13,
                 from Common/Header/StdLinux.h:14,
                 from Common/Header/externs.h:18,
                 from Common/Source/MapDraw/DrawTerrain.cpp:9:
/usr/include/c++/10/type_traits: In substitution of ‘template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Condfalse; _Tp = void]’:
Common/Source/MapDraw/DrawTerrain.cpp:879:5:   required by substitution of ‘template<class Color_t> std::enable_if_t<(sizeof (Color_t) == sizeof (shortt))> TerrainRenderer::drawIsoLinePixel(Color_t*, int16x8_t, uint16x8_t) const [with Color_t = BGRColor]’
Common/Source/MapDraw/DrawTerrain.cpp:968:72:   required from here
/usr/include/c++/10/type_traits:2554:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 2554 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
      |           ^~~~~~~~~~~
Common/Source/MapDraw/DrawTerrain.cpp:890:5: note: candidate: ‘template<class Color_t> std::enable_if_t<(sizeof (Color_t) == sizeof (signed char))> TernRenderer::drawIsoLinePixel(Color_t*, int16x8_t, uint16x8_t) const’
  890 |     drawIsoLinePixel(Color_t *pixel_src, int16x8_t height, uint16x8_t mask) const {
      |     ^~~~~~~~~~~~~~~~
Common/Source/MapDraw/DrawTerrain.cpp:890:5: note:   template argument deduction/substitution failed:
make: *** [Makefile:1714: Bin/LINUX/MapDraw/DrawTerrain.o] Error 1

brunotl, 2022-02-06:
32Bit color was not implemented for OpenGL/Arm.
2 options
 - implement missing function
 - use Open GLES ( add "GLES2=y" to build commande line )

urb31075, 2022-02-06:
Thanks!
**make TARGET=LINUX GLES2=y distrib**
Compiled and run. Everything is working

closed 2022-02-06 by urb31075