LK8000/LK8000 #1575: GDL90 interface

brunotl, 2022-02-06:
protocol spec :
https://www.faa.gov/nextgen/programs/adsb/archival/media/gdl90_public_icd_reva.pdf

Device : 
- Stradux
- SkyEcho
- ?

amagnolo, 2022-04-26:
safesky is an interesting tracking aggregator that has GDL90 output capabilities, so it could be integrated in LK8000 https://www.safesky.app/

Ziozanna, 2023-11-19:
Many people are running safesky that also aggregate flarm asdb amb many others.
It would a great idea to have traffic on lk8000 feeded by safesky and GDL90 protocol.
I personally run safesky in a side phone on every my flight

amagnolo, 2023-11-20:
On the other hand, it could be safesky that implements NMEA out, thus becoming instantly compatible with LK8000 and other popular open source gliding apps. Being a commercial application aiming to expand their reach, they may be interested in pursuing this development.

Ziozanna, 2023-11-21:
Thank you, Alessandro, for your interest. Do you think the NMEA protocol can provide the position of others as GDL90 did?
As far as I know, NMEA is a GPS protocol that manages a single GPS position enriched with other data such as pressure, Xtrack error, etc.
In the event that Safesky sends NMEA, do you believe LK8000 has been designed to display surrounding traffic?
In any case, I am not speaking on behalf of Safesky, but I think that by using GDL90, they have already interfaced with the most popular flying software

amagnolo, 2023-11-22:
I am not an expert on the field, but AFAIK NMEA is used by flarm, thus more popular with gliders, and GDL90 is more used in engine plane avionics. NMEA is fully capable of sending real-time traffic position and alarms (flarm data).
I don't currently used Safesky but it is a good project IMO, they might be well disposed to implement NMEA-out if their customers ask for it. This would greatly expand their reach in the gliding community. @Ziozanna you may ask them... keep us posted if you do. :-)