LK8000/LK8000 #1579: Grand Prix Task - 1.$km (not a typo ;-) - PC Beta Version

JBlyth, 2022-02-25:
I am using LK8000 with Condor 2, and the following has just started happening after upgrading to the latest PC beta version 7.3.10. Having said this, I do not think the issue is anything to do with Condor, as it does not matter if the there is a connection from Condor or not.

I have come across something weird. For Grand Prix tasks ONLY, I get a strange "distance" to the start of the task of 1.$km.

If I edit the task, the issue goes away, and if I cycle through the turn pints, it also reverts to a normal "distance".

I haven't notice this behaviour before. If I change the task to Default, exit, then reopen LK8000, all is normal. If I change the task to GRAND PRIX again, exit, then restart, the issue returns.....

Cheers
John in Sunny Carnarvon, Western Australia
[image]

brunotl, 2022-02-26:
unable to reproduce, Task file and howto replicate is needed .

"1.$" is for "infinite" value ...

JBlyth, 2022-02-26:
Here's the task file. It was originally generated with CoTaCo, which is a great utility for converting Condor 2 tasks to LK8000. However, the task has been saved and re-edited in LK8000, so it should be fully compatible.

Cheers
John

<?xml version="1.0" encoding="utf-8"?>
<lk-task type="Race">
	<options auto-advance="Auto"/>
	<rules>
		<finish fai-height="false" min-height="0"/>
		<start max-height="2449982" max-height-margin="0" max-speed="0" max-speed-margin="0" height-ref="ASL"/>
	</rules>
	<taskpoints>
		<point idx="0" name="Boerne" type="line" radius="3000.000000"/>
		<point idx="1" name="Canyon Lake" type="circle" radius="1000.000000" Exit="false"/>
		<point idx="2" name="Johnson Ranch" type="circle" radius="500.000000" Exit="false"/>
		<point idx="3" name="Boerne" type="line" radius="1500.000000"/>
	</taskpoints>
	<waypoints>
		<point name="Boerne" latitude="29.723801" longitude="-98.694481" altitude="420.000000" flags="5" comment="Alt. limits=0-2450" format="2" freq="123.500" runwayLen="1315" runwayDir="354" style="5"/>
		<point name="Canyon Lake" latitude="29.879900" longitude="-98.241867" altitude="274.000000" flags="1" code="Canyon L" format="2" style="1"/>
		<point name="Johnson Ranch" latitude="30.252201" longitude="-98.622498" altitude="456.000000" flags="5" format="2" freq="123.500" runwayLen="1800" runwayDir="355" style="2"/>
	</waypoints>
</lk-task>

brunotl, 2022-02-26:
lkt file format is other subject. but 2449982 mm = 8037.998 feet, nothing strange.

commit a75ee9af (2022-03-01): PGLineTask optimization : prevent division by 0

fix #1579
closed 2022-03-01 by brunotl