LK8000/LK8000 #1581: LK8000 with Condor VR - Keyboard Control

JBlyth, 2022-04-23:
I see in Version 7.3.12 that one of the improvements was ". Improve keyboard navigation".

Does anyone have any detail on what was changed as keyboard control of LK8000 is quite critical when using Condor 2 in VR?

I have successfully used a number of keyboard commands such as F1, Left/Right, Up/Down and they work well with AutoHotKey and the like, but using MouseClicks for Zoom in/Out and Menu Forward/Back are not really reliable in my experience.

I realise that LK8000 hasn't really been developed for this purpose, but as a long time user of LK8000 and no longer being able to fly in real life, I now "race" with LK8000 almost daily on Condor 2. It's better than not flying at all!

A significant number of young people are being introduced to Gliding through Condor 2. Using LK8000 as a navigation device makes a huge improvement to flying in Condor 2, and of course is the perfect way to practise using LK in a realistic training environment.

amagnolo, 2022-05-30:
Could this be implemented editing a custom "xci" file, like XCRemote does (https://github.com/XCNav/XCRemote/blob/main/XCREMOTELK8000V1.0.xci), to process input from the PC keyboard? Or xci files work only for bluetooth keyboards?

brunotl, 2022-09-18:
closed by  #1597

closed 2022-09-18 by brunotl