LK8000/LK8000 #1590: FTDI4232 (4-port USB to RS232) -- only the first port available on Android platform

henrik1g, 2022-07-12:
Chip datasheet:
https://ftdichip.com/products/ft4232hq/

LK8000 Android Version 7.2.1

Expected behavior

In the device/port dialog all four channels of the FT4232 should show up as separate ports so each can be assigned to an individual driver, so that multiple RS232 devices can be connected.

Actual behavior

In the device/port dialog only the first channel of the FT4232 is available (labeled "USB:0403:6011" ).
This port works as expected. But the remaining three ports do not show up and therefore are unavailable.

Steps to reproduce the behavior

Connect FTDI4232 to USB(-OTG) of an android phone.
I am using this board:
https://www.aliexpress.com/item/1005001928940997.html

When using a serial-terminal app on the android phone, all four ports are available.
I think the driver implementation of this chip in LK8000 does not account for multiple ports.
An update of the driver code to assign all available ports for this chip would resolve the issue.

I am of course available for testing new code since I have the hardware here.

Side note:
Using this: USB-C-(OTG)-PD adapter
https://de.aliexpress.com/item/1005003887696758.html
the (PD-enabled) phone can get reliably charged while acting as USB-host for the FT4232, so this is a viable setup and will be interesting for many people.
(The phone needs to be capable of USB-C-PD (PowerDeliveryProtocol) though)
Tested on Pixel4a.

brunotl, 2022-07-12:
same problem with two identical USB device connected was fixed by this commit : d8cdbbde75fe543957e7a2377b473890a35a0875

henrik1g, 2022-07-12:
I would like to test if that commit also solves the issue for FT4232.
I think this commit is not yet included in the 7.3.17 (beta) from play store.
Is there a place where I can get a nightly apk from actual master branch?
Or when will the beta on play store be updated?

brunotl, 2022-07-13:
looking FT4232 datasheet, problem is not solved by commit d8cdbbd.

lyusupov, 2022-07-14:
According to [this site](https://github.com/linuxhw/LsUSB), the FTDI 4232 is 1 high-speed device with 4 interfaces and each of them has 2 endpoints (IN and OUT):

```
Bus 001 Device 004: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6011 FT4232H Quad HS USB-UART/FIFO IC
  bcdDevice            8.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  --
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          101
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

henrik1g, 2022-07-21:
Looking at the code it seems to me that the solution needs to start here:
https://github.com/LK8000/LK8000/blob/66a1fd6d61c82efa1997ed7e5fe49afa0cea6fd8/android/src/org/LK8000/UsbSerialPort.java#L48
The call to the overloaded function createUsbSerialDevice() omits the option for the index of the interface, which makes it default to interface 0.

See here for the function including the interface number:
https://github.com/felHR85/UsbSerial/blob/7fff8b6d5ca19590dcb05c3f977970e8cce103b7/usbserial/src/main/java/com/felhr/usbserial/UsbSerialDevice.java#L66

henrik1g, 2022-08-20:
With the help of the XCSoar community I was able to fix the issue in XCSoar.
Maybe you also want to pull this into LK8000.
https://github.com/XCSoar/XCSoar/pull/995

brunotl, 2022-08-21:
I've already made same kind of change to allow to use device with multiple interface, but without device i can't test before merging to master ...

https://github.com/brunotl/LK8000/commit/17d076fb2dae4318a6306bb25f2e4996002c259b

lyusupov, 2022-11-27:
@brunotl 
If we will take a look on GitHub activity of this user https://github.com/henrik1g
we can find out that 
* he came out from nowhere after a years of silence
* made a tweak of XCSoar to achieve some goal that nobody else needs except him
* have caused a regression of XCSoar USB function https://github.com/XCSoar/XCSoar/pull/995#issuecomment-1308471377
* disappeared from GitHub and takes no responsibility for the fault he did

Thus, IMHO, it has little sense to keep this ticket opened. We are safe to close it with 'no response' resolution.