LK8000/LK8000 #1591: OpenStreetMaps or similar

andrebstv, 2022-07-19:
It would be really nice if we could use community maps like openstreetmaps or mapsforge maps on LK8000. I'm a real fan of the app, but the maps are very poor. 

Is the code base somewhat compatible? I would put in the effort to make this update if it was. And also if you all agree that's a nice feature.

brunotl, 2022-07-24:
what is your things behind "Is the code base somewhat compatible" ?

brunotl, 2022-07-28:
maplibre-gl-native library look compatible with LK8000 project but only for OpenGL target ( Android / Linux ), that exclude Kobo & Windows platform.

https://github.com/maplibre/maplibre-gl-native

andrebstv, 2022-08-11:
@brunotl is there a discord / messaging group where we can discuss the idea better ?

brunotl, 2022-08-14:
@andrebstv We have Slack channel, send PM to me ...

andrebstv, 2022-08-24:
Can you send me the link ? I've never used slack =]

brunotl, 2022-08-24:
email is needed to add you in slack workspace

andrebstv, 2022-08-25:
***@***.***

brunotl, 2022-08-25:
you can contact me with : bruno@lk8000.it