LK8000/LK8000 #1592: Terrain Format, documentation or howto?

lordfolken, 2022-07-29:
Hi, 

I have full planet srtm data here on https://mapgen-data.sigkill.ch/dem3/
I'd love to be able to generate maps for LK8000 with XCSoar's mapgen.

The formats for topology and so are nearly, the same, however terrain seems to be nothing that my normal gis tools can interpret. 

Is there a howto or script of how to generate the DEM file?

brunotl, 2022-07-29:
simple c# fonction  to convert  ESRI ASC to DEM 

https://github.com/ToninoTarsi/GIS2LKMAP/blob/39bd5bca99b82dc77b5b50ba37ab61efb2592c7e/GIS2LKMAP/Form1.cs#L95

JBlyth, 2022-11-03:
I have almost finished a "GenerateLKDem" app that will quickly download an .ASC file and format it as a .LKM from either the command line or the app itself (utilising Tonino's C# routine). I'll look at putting it onto GitHub in a few days after I have finished some refinements.

If I can get a method to do the .DEM file as well, I would be happy to release it. Though it won't be as graphically flash as "LKMap Desktop"....

I would be very happy to collaborate on producing something that would produce both .DEM and .LKM files for LK8000 Users.

Cheers
John - in Sunny Carnarvon, Western Australia

JBlyth, 2023-05-12:
Hello Andrebstv. The really good news is that LKMaps Desktop Version 2 (the son of Version 1.x ;-) should be available very soon.

Watch this space!!!!!