LK8000/LK8000 #1596: Could you add support for a VR/PC Keyboard please? The code snippets below seem to work nicely.

JBlyth, 2022-09-16:
I have had a very interesting time setting up Linux on my Win11 Laptop so I can compile LK8000, and then delving into the mysteries of how beautifully LK8000 is written and structured - a big learning curve yesterday on both counts!

I think the additions below will give users an easier method to use LK8000 with Condor 2, particularly in Virtual Reality or when using LK8000 on a second screen. I use it in conjunction with AutoHotKey to bring LK8000 to the foreground as required so the keypresses are pushed to control LK almost fully with my Warthog Joystick and HOTAS. I can also hold LK in the foreground for more complex tasks and use a mouse, again with AutoHotKey .

My suggested inclusions in LK allow a normal PC keyboard to control all the screen clicks currently available, plus all ten possible Custom Menu Items.

I have been able to nut out how to add an extra keyboard in LK8000 by:-
Adding at line 36 in **\LK8000-master\Common\Source\Models.cpp**

        { PCKEYBOARD, _T("PC KEYBOARD") },

AND adding at line 49 in **\LK8000-master\Common\Header\Modeltype.h** 

        PCKEYBOARD   = 10025, // VR-PC Keyboard

AND adding at line 1725 in **\LK8000-master\Common\Source\Draw\MapWndProc.cpp**

    //
    // Virtual Reality/PC Keyboard type 1.
    //
    if (ModelType::Get() == ModelType::PCKEYBOARD) {
        #ifndef __linux__
        #if (WINDOWSPC<1)
        if (!Debounce(KEYDEBOUNCE)) return;
        #endif
        #endif
        switch (KeyCode) {
                // THE BOTTOM BAR
             case KEY_Q:
                 key_bottombar_previous();
                return;

            case KEY_W:
                key_bottombar_next();
                return;

            case KEY_E:
                key_previous_mode();
                return;

            case KEY_R:
                key_next_mode();
                return;

                // THE LEFT SCREEN
            case KEY_T:
                key_previous_page();
                return;

                // THE RIGHT SCREEN
            case KEY_Y: 
                key_next_page();
                return;

                // THE CENTER-UP
            case KEY_U:
                 key_up();
                return;

                // THE CENTER-DOWN
            case KEY_I:
                key_down();
                return;

                // THE TOP-LEFT
             case KEY_O:
                key_topleft();
                return;

                // THE TOP-RIGHT
            case KEY_P:
                key_topright();
                return;

                // ENTER
			// case KEY_[:
            case 0x5b: // VK_OEM_[
                key_enter();
                return;
 
				// OVERTARGET ROTATE (WHAT DOES IT DO?)
			//case KEY_]:
			case 0x5d: // VK_OEM_]
                key_overtarget_rotate();
                return;

			case KEY_A: // Custom Key 1
				CustomKeyHandler(CustomMenu1 + 1000);
				return;
			case KEY_S: // Custom Key 2
				CustomKeyHandler(CustomMenu2 + 1000);
				return;
			case KEY_D: // Custom Key 3
				CustomKeyHandler(CustomMenu3 + 1000);
				return;
			case KEY_F: // Custom Key 4
				CustomKeyHandler(CustomMenu4 + 1000);
				return;
			case KEY_G: // Custom Key 5
				CustomKeyHandler(CustomMenu5 + 1000);
				return;
			case KEY_H: // Custom Key 6
				CustomKeyHandler(CustomMenu6 + 1000);
				return;
			case KEY_J: // Custom Key 7
				CustomKeyHandler(CustomMenu7 + 1000);
				return;
			case KEY_K: // Custom Key 8
				CustomKeyHandler(CustomMenu8 + 1000);
				return;
			case KEY_L: // Custom Key 9
				CustomKeyHandler(CustomMenu9 + 1000);
				return;
			case KEY_Z: // Custom Key 10
				CustomKeyHandler(CustomMenu10 + 1000);
				return;
 
			//case KEY_X: // BottomCentre (Taken from XCREMOTE - but doesn't do a bottom bar centre click
			//	key_next_mode(); // bottom bar center click
			//	return;
			default:
				break;
        }
    }
    //

amagnolo, 2022-09-16:
Excellent work JBlyth, thank you for sharing.

You could add a little bit of "git" to your learning experience and make a pull request (PR) - that is, committing these changes you are proposing in a more formal way so they can be  checked and approved by a maintainer. 
If you do the PR yourself, it will be faster and easier for them to integrate your change, and you could more easily make changes or improvements in the future.

Another suggestion is to write some documentation for end users regarding how to use this new feature (it should go in the Docs folder, maybe in the manual but that needs some maintenance too).

This solves also issue https://github.com/LK8000/LK8000/issues/1581

brunotl, 2022-09-16:
Duplicate of #1581

brunotl, 2022-09-16:
- your key binding look like a "QWERTY" keyboard, not a "PC" keyboard, 
- using hex keycode are not cross platform : cf [Common/Source/xcs/Event/Key.h](https://github.com/LK8000/LK8000/blob/84295fe3ebaa24012b6250498459122143084dc5/Common/Source/xcs/Event/Key.h) for kyecode
- as @amagnolo say, instead of open new issues, do  PR. cf [Github doc](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork)

JBlyth, 2022-09-16:
Ok. I will have a look at adding some "git" to my diet ;-)

You are right of course, Brunotl. A Qwerty keyboard. I took the idea of hex codes from the current BTK 1-3 etc keyboards, but I will review the Key.h keycodes and try and do better next time :-)

JBlyth, 2022-09-17:
I'm looking at creating a pull request. Which branch do I use? Master or 7.2? Sorry this is my first experience at this GitHub stuff, so I wish to tread carefully.

I have modified my script changes to now call the keyboard type "QWERTY" and "QWERTY KBRD" in the Devices dialogue, and removed the hex numbers. The 10 Custom Menu items now run Q to P, the screen position keys A to H, and also the Up/Down/Left/Right keys. 

I reviewed the key.h files, and although somewhat confused, I hope I am now on the right path.

Once I have things moving in the right direction, I am happy to add to the Docs folder on using LK8000 with Condor 2 in Virtual Reality or in a multi-monitor setup - which has caused a lot of angst for a lot of people, along with descriptions of the methods used and sample AutoHotKey scripts if desired. 

I can also prepare something shorter for the Manual along similar lines to how the current BTK Keyboards are described if you wish....

JBlyth, 2022-09-17:
I now have these two entries. Am I on the right path?

[image]

and 

[image]

Obviously I should have combined these somehow. Sorry to be asking these questions....

Please see the above post as well.

brunotl, 2022-09-17:
maybe you need to start by reading that :
https://gist.github.com/Chaser324/ce0505fbed06b947d962

JBlyth, 2022-09-18:
Well, I have spent/wasted hours going through every single command line in that document. Was it missing the bits about adding files and "committing" perhaps? 

I've got as far as shown below is the output (edited) of the terminal window. It all seems to be far more complicated that it needs to be. I've spent vastly more time on trying to get it into GitHub via Linux that it took me to figure how to make the changes in the LK8000 software itself, but the "upload" to the Github FORK hasn't done anything it would seem. No doubt I have missed some stage somewhere, but I really need someone to turn on the light at the end of the tunnel for me, otherwise I may as well give up, I'm afraid ....

I have no idea why some of the lines below are struck out. I pasted them straight from Notepad.exe

15P:~/LK8000$ git status
On branch Add-Qwerty-Keyboard
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   Common/Header/Modeltype.h
        modified:   Common/Source/Draw/MapWndProc.cpp
        modified:   Common/Source/Models.cpp

no changes added to commit (use "git add" and/or "git commit -a")
15P:~/LK8000$ git add Common/Header/Modeltype.h
15P:~/LK8000$ git add Common/Source/Draw/MapWndProc.cpp
15P:~/LK8000$ git add Common/Source/Models.cpp
15P:~/LK8000$ git status
On branch Add-Qwerty-Keyboard
Changes to be committed:
  (use "git restore --staged <file>..." to unstage)
        modified:   Common/Header/Modeltype.h
        modified:   Common/Source/Draw/MapWndProc.cpp
        modified:   Common/Source/Models.cpp

15P:~/LK8000$ git merge upstream/master
Already up to date.
15P:~/LK8000$ git checkout Add-Qwerty-Keyboard
M       Common/Header/Modeltype.h
M       Common/Source/Draw/MapWndProc.cpp
M       Common/Source/Models.cpp
Switched to branch 'Add-Qwerty-Keyboard'
15P:~/LK8000$ git rebase master
error: cannot rebase: Your index contains uncommitted changes.
error: Please commit or stash them.
15P:~/LK8000$ git commit --help
15P:~/LK8000$ git commit
Aborting commit due to empty commit message.
15P:~/LK8000$ git commit
[Add-Qwerty-Keyboard 56e5641a5]  On branch Add-Qwerty-Keyboard  Changes to be committed:        modified:   Common/Header/Modeltype.h   modified:   Common/Source/Draw/MapWndProc.cpp      modified:   Common/Source/Models.cpp
 3 files changed, 110 insertions(+)
15P:~/LK8000$ git commit
On branch Add-Qwerty-Keyboard
nothing to commit, working tree clean
15P:~/LK8000$ git commit -a -m "Add QWERTY Keyboard for VR and PC Use"
On branch Add-Qwerty-Keyboard
nothing to commit, working tree clean
15P:~/LK8000$ git checkout newfeature
error: pathspec 'newfeature' did not match any file(s) known to git
15P:~/LK8000$ git checkout Add-Qwerty-Keyboard
Already on 'Add-Qwerty-Keyboard'
15P:~/LK8000$ git rebase master
Current branch Add-Qwerty-Keyboard is up to date.
15P:~/LK8000$ git checkout
15P:~/LK8000$ git rebase -i master
Successfully rebased and updated refs/heads/Add-Qwerty-Keyboard.
15P:~/LK8000$

brunotl, 2022-09-18:
git is in fact pretty simple when you know what is "upstream" , "origin" , "local", "index" and "working dir" and what is the command  "pull" "checkout", "add" "commit", "push",  most of the time this command is only in this order ...

"pull" -> to get change from origin or upstream
"checkout" -> to switch from branch
"add"  -> to push woking directory change to "index" (staged change)
"commit" -> to commit staged change to local
"push" -> to uplaod local commit to "origin" ( your fork )

maybe the document i've sent is not the best, but it's start point, web is full of howto to explain the right way to use git, starting by the official git website https://git-scm.com/doc

brunotl, 2022-09-18:
closed by  #1597

closed 2022-09-18 by brunotl