LK8000/LK8000 #1605: TMA airspaces not recognized

amagnolo, 2023-02-10:
Airspaces of type "TMA" from openaip database are not recognized. This is particularly dangerous for example around Locarno CTR, where there are 6 TMAs that may not be reported by LK8000.

In openair format they are identified by `AC TMA`; LK8000 loads them as "Unknown" type. 
Sample: [ch_asp.txt](https://github.com/LK8000/LK8000/files/10709835/ch_asp.txt)

In aip format they are identified by `<ASP CATEGORY="TMA">`; LK8000 doesn't load them at all.
Sample: [openaip-airspaces(ch).txt](https://github.com/LK8000/LK8000/files/10709855/openaip-airspaces.ch.txt) (the file must be renamed with extension .aip to be recognized by LK8000)

brunotl, 2023-02-11:
"TMA"  is not an airspace class, in your sample ch_asp.txt , some TMA airspaces are class "D" ( TMA ALPNACH 2, TMA ALPNACH 4... )  some others are class "C" ( TMA BALE ZURICH AZ3 )

your airpsaces data source is invalid ...

only glitch is the difference between openair/openaip unknown airspaces handling.

closed 2023-02-12 by brunotl

amagnolo, 2023-02-12:
You are right, the data source may not be reliable.
So I went straight to openaip.net and downloaded Switzerland airspaces in all the formats supported by LK8000.

Here I am assuming that openaip.net is to be considered a reliable source for use in LK8000; if it is not, please disregard the rest and point me to what should be considered the reference airspaces to use (the ones on LK8000 website are way too old). 

I am focusing on TMA LOCARNO 1 to 6:

* The file in openair format sets "AC TMA", loaded as "Unknown" in LK8000 [openaip-ch.openair.txt](https://github.com/LK8000/LK8000/files/10715866/openaip-ch.openair.txt)

* The file in openair extended format sets "AC D", so this seems to be the most correct one, but gives parse errors when loaded in LK8000 [openaip-ch.openair-extended.txt](https://github.com/LK8000/LK8000/files/10715870/openaip-ch.openair-extended.txt)

* The file in openaip format sets ASP CATEGORY="TMA", currently not loaded by LK8000 but I guess it will be "Unknown" with your patch [openaip-ch.openaip.aip.txt](https://github.com/LK8000/LK8000/files/10715876/openaip-ch.openaip.aip.txt)

Would it be possible to make LK8000 compatible with openaip.net files, even tough (as they state in the header) they "may contain non-standard AC values" ? 

At least for the  openaip format, where I think they are the ones who define the standard.

amagnolo, 2023-02-26:
At the moment the workaround is to use maps from https://www.openflightmaps.org/