LK8000/LK8000 #1607: 7.4.1. Optimization for cylinders over 100 km does not work

mymyser123, 2023-02-19:
Optimization for cylinders over 100 km does not work. The optimization line goes not to the border of the cylinder, but to its center, for example, if you are already inside the cylinder at the previous TP and you need to fly to the exit .. The total length of the distance is also incorrectly calculated.

brunotl, 2023-02-19:
Please add the task file to the ticket...

brunotl, 2023-02-20:
tested, work perfect ...

[image]

closed 2023-02-20 by brunotl

mymyser123, 2023-02-21:
[TASK-EX.zip](https://github.com/LK8000/LK8000/files/10790805/TASK-EX.zip)
[image]
[image]

brunotl, 2023-02-21:
when task optimizer is not disabled : 
[image]

that work perfectly
[image]