LK8000/LK8000 #1608: Title display (ID) Turnpoints Name

mymyser123, 2023-02-19:
If the TP short code (like D01) is in the "ID" field and not the "Name", then it is not displayed. Please make a button to switch between the display modes of the "ID" and "Name" fields in the point file. How is it done in Xctrack/SeeYou.

brunotl, 2023-02-19:
Please add turnpoint file and config to reproduce.

brunotl, 2023-02-19:
if you want to display ID instead of name go to "system config Page "3 Map Display" and choose "ICAO code" instead of "Names" for "Labels"

mymyser123, 2023-02-21:
This problem in displaying when task is seting. There, only the name field is always displayed.
[Waypoints file](https://airtribune.fra1.digitaloceanspaces.com/media/contest/waypoints/2022/01/openroldanillo2022.SeeYou.cup)

[image]
[image]