LK8000/LK8000 #1609: ESP32 Bluetooth SPP does not connect

andrebstv, 2023-02-26:
Using a Bluetooth serial device made by a ESP32 device.

LK8000 (on android) does correctly identifies and find the device. Though it receives no data. Using the classic Bluetooth Serial Monitor app, available at the playstore, confirms that the ESP32 based device is transmiting data properly.

andrebstv, 2023-02-26:
Do notice the error saying that it`s not possible to open the comm port. Maybe permissions ?

andrebstv, 2023-02-26:
> Using a Bluetooth serial device made by a ESP32 device.
> 
> LK8000 (on android) does correctly identifies and find the device. Though it receives no data. Using the classic Bluetooth Serial Monitor app, available at the playstore, confirms that the ESP32 based device is transmiting data properly.

Checking the latest LK8000 beta manifest it caps the permission at Android 11 (Sdk version 30). Maybe that`s the problem.

brunotl, 2023-02-26:
only BLUETOOTH_CONNECT and BLUETOOTH_SCAN are required since android 12 : 
https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android12-or-higher

brunotl, 2023-03-04:
Enable to reproduce ...

brunotl, 2023-03-26:
fixed by commit : 8d24d568f62c2e6c84fa43022745ad886363fc1f

closed 2023-03-26 by brunotl