LK8000/LK8000 #1611: Add SkyDrop2 bluetooth device

fhorinek, 2023-04-12:
My name is Fero and I am a developer from SkyBean. Our vario SkyDrop worked with your app when using a classic Bluetooth SPP profile.

During the chip shortages, the old module and also other SkyDrop components became unavailable, therefore we needed to change them.
Bluetooth on SkyDrop2 is HM-BT4502 and it does only support BLE.

Can you please add our device to your application?

The parameters for BLE are:

Service UUID
0000ffe0-0000-1000-8000-00805f9b34fb
Read characteristic UUID
0000ffe4-0000-1000-8000-00805f9b34fb
Write characteristic UUID
0000ffe9-0000-1000-8000-00805f9b34fb

Normally we transfer telemetry using LK8EX1, but we can also emulate DigiFly or BlueFly (selectable in the device).
We can also forward GPS data using standard NMEA sentences (if enabled).

If you need to test it on a device, let me know, and we can provide one.

brunotl, 2023-04-12:
Hi Fero 

yes it's possible to add it, and yes device is needed for that...

2 questions : 
 - why do not use HM10 read/write characteristic UUID ?
 - what BLE mtu size do you use for characteritics ?

ElDevelish, 2025-03-29:
Hi,
would that also fix the non-working state of BLE telemetry for the SkyBean Strato?

brunotl, 2025-05-23:
Done, but untested due to lack of device.

closed 2025-05-23 by brunotl