LK8000/LK8000 #1616: KOBO GLO HD Screen modes turn of 90 degrees freeze shutdown

Ziozanna, 2023-05-09:
Good morning everyone.
I found a problem in this release and in 744 beta as well on Kobo glo HD.
If I change the display mode of 180dgr is fine but if I change it of 90dgr once I turn LK8000 off it doesn't shut down properly , the sceen freeze on saving logs and at the next reboot it starts on KOBO OS.
Found a workaround. If you change the screen orientation without fly before shutdown LK8000 stuck on writing logs maybe beause there is nothing to log.
I entered into a simulation mode and made a short fly then boom. Correct shutdown is performed and the screen was swapped to 90degr permanently

Ziozanna, 2023-06-04:
This is the same bug that freeze the KOBO on writing logs during auto landing detection or shutdown

brunotl, 2023-06-06:
fixed by commit 5022b46

closed 2023-06-06 by brunotl