LK8000/LK8000 #1620: IGC logs are no longer accessible

krzysztof-lopacki, 2023-06-18:
After an upgrade to a new file system introduced in Android 11 the LK8000 saves IGC files in this location:
`/storage/emulated/0/Android/data/org.lk8000.beta/files`
(which points to the same resource as `/sdcard/Android/data/org.lk8000.beta/files`).

The problem is that this directory is NOT accessible to:
- any files manager installed on the phone
- IGC Viewer
- Windows file explorer when a device is connected to PC.

The only way to access the logs is through ADB or Android Studio (which uses ADB under the hood).
This is a real pain on the airfield that after the flight I cannot review it, check the start/landing time etc. I need to connect the phone to my dev machine to do this.

Possible solution:
Add "Copy IGC Logs to Download" button which copies all the stored logs to `sdcard/Download/lk8000/_Logger` directory (or any other that is publicly accessible).

amagnolo, 2023-06-18:
Some file managers (i.e. [X-plore](https://play.google.com/store/apps/details?id=com.lonelycatgames.Xplore)) can access that directory, no root required

brunotl, 2023-06-18:
files are available from android file manager and any app allowing to select files using android UI :
[image]

closed 2023-06-18 by brunotl

bambukouk, 2023-06-18:
Using IGC viewer I can open old logger  files but not recent?
Is this IGC viewer problem? or is there something still not working as expected, please?
or is it me being stupid again :+1:  ?

brunotl, 2023-06-19:
IGC Viewer is dead app, not updated since 2018 and no more available on GPlay probably for lack of recent android version compatibility ...

bambukouk, 2023-06-19:
@brunotl  thanks - you are probably right about IGC viewer, which is the shame because it is a nice app :+1: 
and I don't know of similar one... for use with my Android phone.
What I particularly liked about it was that you could view directly LK8000 logs.
Now I have to download them to view elsewhere :-1:

brunotl, 2023-06-19:
take look to this : https://xcanalytics.fr

bambukouk, 2023-06-19:
@brunotl 
thank you for suggesting an alternative 👍
annoying thing is that IGCviewer still works fine with XCSoar 😛

brunotl, 2023-06-20:
> @brunotl thank you for suggesting an alternative +1 annoying thing is that IGCviewer still works fine with XCSoar stuck_out_tongue

that depends of android version and where is the prf file you have select at startup...

krzysztof-lopacki, 2023-06-20:
I confirm that [X-plore](https://play.google.com/store/apps/details?id=com.lonelycatgames.Xplore) can access the files in the `/sdcard/Android/data/org.lk8000.beta/files` directory. It is enough to copy the files from `_Logger` to publicly accessible folder (e.g. Downloads) so the IGC Viewer can access them.

Unfortunately a few files managers I've tested were not able to do the trick (they can't utilize ACTION_OPEN_DOCUMENT_TREE intent). Even the build in file manager provided by Oppo cannot open LK8000's directory.

Anyway, problem solved. Thank you @amagnolo and @brunotl for the tips!

brunotl, 2023-06-20:
some manufacturer remove the stock android file mananger shortcut ...

to restore this shortcut you can install this : https://play.google.com/store/apps/details?id=org.aospstudio.files&hl=fr&gl=US