LK8000/LK8000 #1628: Mediterranean Islands "disappear" in PC LK8000

JBlyth, 2023-08-06:
As discovered by Alf, it appears that the islands of Corsica, Sardinia, and Sicily "disappear" form the LK8000 map when using the PC version. They magically reappear when using the Android version.

In both instances version 7.4.7 is being used on both devices. The .DEM, .LKM, and Config files are the same on each device.

I have uploaded a PC version screenshot and an Android version screenshot that demonstrates the issue to the LK8000 "Bugs" forum.
https://www.postfrontal.com/forum/topic.asp?TOPIC_ID=9406

The Topology seems to be present when I zoom into where Sicily should be, but there are no elevations shown in the PC version, but is obviously present in the Android version.

Cheers,
John in Sunny Carnarvon - Western Australia

brunotl, 2023-08-06:
please join link to dem/lkm files to replicate...

brunotl, 2023-08-16:
not LK8000 bug, it's KLMaps_Destop  bug ...

closed 2023-08-16 by brunotl

JBlyth, 2023-11-05:
Beautifully fixed. Thank you for all your hard work, Brunotl :-)