LK8000/LK8000 #1629: Kobo freeze on reconnect serial ports

Ziozanna, 2023-10-29:
Good morning,

My Kobo GLO HD is connected to a GPS and a serial adapter via OTG using a small USB hub. The OTG connection is a Y-cable that supplies power to both run the Kobo and external devices.

Sometimes, external power fails for various reasons. When the power is restored, my Kobo freezes. The only way to restart it is by performing a hard power cycle on the Kobo.

Thanks to take this behavior into account

Andrea

brunotl, 2023-10-29:
what LK8000 version ? 

have you a core dump files inside folder "/LK8000/kobo/crash/" ?

Ziozanna, 2023-10-30:
By tomorrow afternoon I'll check .

Thanks Bruno

Ziozanna, 2023-10-31:
[inbound5983130836863118426.zip](https://github.com/LK8000/LK8000/files/13219931/inbound5983130836863118426.zip)
Ciao Bruno

Here the coredumps and a couple of video to show when kobo crash

This video did not crash at the first time but only when I turn on bluefly

https://youtu.be/MbF-ZtiAWAM?si=sB8wVgWwVR1FXsSM

In the following I did turn on bluefly at startup and kobo crash just after 1 St time power restored

Maybe you did not recognize me.
I'm the same Andrea of the flybox

Cheers

brunotl, 2023-11-06:
fixed by commit b1bbbbb9

closed 2023-11-06 by brunotl