LK8000/LK8000 #1631: FLARM

doramail, 2023-11-08:
Hi,

When I launch LK8000 version 7.4.5 (not beta), in simulation mode I have FLARM activated but not with latest beta version 7.4.9.
Is there an issue ?
Best regards

brunotl, 2023-11-08:
what platform ?

check if traffic on map is enabled ( system config page "13 Map Overlay" )

brunotl, 2023-11-08:
Flarm SIM is also desable if  "Enable Radar" = ON in Pilot config ...

doramail, 2023-11-08:
Yes, it is set to ON on both versions, same configuration but different behavior, in release version, I can see the Flarm page but not with beta version.

doramail, 2023-11-08:
Sorry I didn't understantd at first read.
For the Flarm to display in SIM Mode, "Enable Radar" has to be "OFF" in Pilot Configuration and traffic on map is enabled ( system config page "13 Map Overlay" ).
It works now on both versions, Thank you very much.

closed 2023-11-08 by brunotl