LK8000/LK8000 #1646: Condor 3 device requested to correct wind being out by 180 degrees

JBlyth, 2024-11-19:
I have been a beta over past months for Condor 3. The development team has discovered an error in the NMEA stream for wind in earlier Condor versions, where the wind data is out by 180 degrees. With LK8000, you have corrected this error in the Condor device by adding180 added to the wind calculation to correct that error in the Condor NMEA output string.

Line 115 from https://github.com/LK8000/LK8000/blob/master/Common/Source/Devices/devCondor.cpp reads:-
 double wfrom = StrToDouble(ctemp,NULL) + **180**;

In Condor 3, this error has been corrected in the interests of accuracy, so it would be very helpful if LK8000 could be updated to resolve this change for Condor 3 users. 

It may be easiest to just add a Condor 3 device to the list with that line changed to:
double wfrom = StrToDouble(ctemp,NULL) ;

That would maintain compatibility with Condor 2 while giving Condor 3 users the correct wind data.

brunotl, 2024-11-19:
try the "LX"  driver, Condor 3 seem's to send correct "LXWPx" stream so it should work ...

JBlyth, 2024-11-22:
While the LX Device does work, the refresh rate is only once every 4 seconds or so with LK8000.

brunotl, 2024-11-22:
can you share a NMEA log fragment ?

JBlyth, 2024-11-22:
With the Condor device or the LX device in LK8000, or the IGC output from Condor 3 (which may not have the wind data)?

witorwitor, 2024-11-22:
here is a bit of NMEA from Condor 3:

$LXWP0,Y,118.5,1760.8,-1.07,,,,,,180,**180**,25.0*63 - Correct NMEA, shows wrong direction on LK8000. The direction should be 000 not 180

brunotl, 2024-11-22:
> While the LX Device does work, the refresh rate is only once every 4 seconds or so with LK8000 and it seems there are other issues as well.

look like invalid  RMC and/or GGA sentence ...

if someone can provide few minute minute of NMEA log, i can investigate.

in all case copying the wall current Condo device to just remove "+180" offset on wind bearing is not the right solution to solve the issues because each time code is copied, we can be sure some copy will become a bug in near future ...

witorwitor, 2024-11-22:
not sure if this is what you are after, but here is my NMEA log from the last flight in Condor 3
[NMEAlog.txt](https://github.com/user-attachments/files/17874965/NMEAlog.txt)

lordfolken, 2024-11-24:
@brunotl i opted for a bool in the LXWP parser: 
https://github.com/XCSoar/XCSoar/pull/1614/commits/452771d112fa1a00eb2483056dc1faba9725ad4a

closed 2024-11-24 by brunotl

JBlyth, 2024-11-25:
Thank you. That will be great for all those Condor 3 pilots out there!