LK8000/LK8000 #1650: When will a new version be released to include the Condor 3 Wind Reversal update?

JBlyth, 2025-01-08:
I am desperately hanging out for a new release which includes the FIX for the Condor 3 Wind Reversal issue that you so neatly fixed back in November. Any chance of releasing it soon? :-)

[https://github.com/LK8000/LK8000/issues/1646]

witorwitor, 2025-02-01:
So am I and many more people who want to practice using LK8000 while simming in Condor3. It is winter in Europe so best time to practice in a simulator. Please, please, please release a version with the corrected wind direction...

brunotl, 2025-02-04:
planned for this week

JBlyth, 2025-02-04:
Ah, you're an ABSOLUTE GEM!

Thank you!

closed 2025-02-16 by brunotl

witorwitor, 2025-02-17:
Very useful for me. Nice to see live wind on LK. Thank you for implementing.