LK8000/LK8000 #1651: Multi turn-point tasks

gwirdnam, 2025-02-08:
Thanks for the hard work on LK8000 beta. It's still the best tool for comp and free flying in my opinion.
I recently flew some tasks that were multi turnpoint tasks in a small area, as many as 12 turnpoints plus start/finish cylinders in similar or identical positions. It was difficult to keep track! The map was useful but very busy with crossing lines and circles, the most useful thing was the different coloured line for the current leg and cylinder. But it would have been also useful to have a different coloured cylinder for the cylinder that came next.
As a suggestion I would say use a bright colour for the next cylinder/track line and a medium bright colour for the one after that and the subsequent cylinders and lines should be muted further so they don't clutter the display too much. This would have the effect of a "visual priority" for the turnpoint order. 
I attach an example of one of the tasks we flew. GitHub won't allow a .LKT file as an attachment so it's attached as a text file instead.

[T5.txt](https://github.com/user-attachments/files/18718840/T5.txt)

[image]

Thanks, Gary

brunotl, 2025-05-23:
Could you suggest a set of colors or a color progression you think would work well. For example:

- What **bright color** would you suggest for the *next* cylinder?
- What would be a good **medium-bright** color for the one *after* that?
- Do you have any preferences for **muted tones** for the later turnpoints?
- Should the **line colors** match the cylinder outlines?

gwirdnam, 2025-05-25:
I would go for something like:

Start/Finish cylinder : red

Target cylinder/track towards : magenta

Next cylinder/track : bright green

Subsequent cylinders/tracks : dark green/brown colour

Gary