LK8000/LK8000 #1655: Auto force final glide before last waypoint

Rogerwii, 2025-06-07:
Hello, 

I have noticed that final glide mode will only activate when the target is the last waypoint of the task (goal). It should trigger at least on the second last waypoint (end of speed section, often 600m closer) and probably as soon as the required altitude to reach goal is reached!

For now, even in auto Thermal + Final mode, McCready stays on thermal even when above final glide to goal.

Is this a bug?

Thanks in advance.

brunotl, 2025-06-09:
Should switch to final glide if arrival height is upper than 120m if "Auto Final Glide" is set to ON (System config "5 Glide Computer"

Kamaltdinov, 2025-10-16:
I conducted two experiments related to this commit, which was later reverted. It turned out that it works (with a minor caveat) in scenario (case1), but does not work in (case2). Here’s how it happens:

**Case1 Slovenia** ([files for reproduction](https://drive.google.com/file/d/1cS5geYCpAbIXN1I9cp06nc9c3Un6aS4B/view?usp=sharing)):

1. While flying in the final glide mode, as soon as the arrival height at the finish exceeded 120m, I saw the message "above final glide" and the letter next to the MC changed to "F".
2. However, I had not yet reached the last TP, since it is located high in the mountains while the finish is in the valley.
3. The MC value itself remained the same as in the "Thermal" mode and did not change, even though the finish altitude changed.
[image]

5. As soon as I approached (but did not yet reach) the TP and the glide to it became positive, the MC value suddenly jumped from the last thermal value to the actual final glide value.
6. Thus, everything worked correctly except that, in my opinion, the message and mode switch occurred prematurely.  

I suggest checking not only the glide to the finish but also the glide to all remaining TPs, if possible. Otherwise, the pilot may be misled: they see "F" and think they can reach the finish, stop searching for thermals, while in fact, they are still in "Thermal" mode since they haven’t reached the TP, even though they have reached the finish.

**Case2** ([files for reproduction](https://drive.google.com/file/d/1iZJw11PzBUdRQcnWi9sDJ_tVXh-B51Vy/view?usp=sharing)):
Things are significantly worse here. As I described [here](https://github.com/LK8000/LK8000/issues/1662), when reaching the final glide range, the readings start to fluctuate and the mode switches between Thermal and Finish every 1-2 seconds.

I don’t understand the difference from the first scenario, except that the TP requires a higher altitude than the finish (unlike case1).

Kamaltdinov, 2025-10-16:
I managed to figure out the difference. 

In Сase1, if you simply move the last TP lower down the slope to make it more easily achievable than the finish point, the mode switching starts immediately. This means that when the finish is reachable but the TP is not, it somehow blocks the "oscillations".

However, if you set the MC to final glide mode instead of final+cruise, its value is calculated to reach exactly the TP, not the finish point as we would expect. In other words, the arrival altitude at the TP is estimated at around 0, while the arrival altitude at the finish is, for example, -300m.

I'm not an expert in the algorithm, so I tried analyzing it with AI and here's what I found:

**The key bug** - `PirkerAnalysis` doesn't account for the entire task during forced final glide in multi-leg tasks. It only works correctly on the final leg (single-leg). In multi-leg scenarios, this leads to overestimation of available MC, "overspending" of altitude, and resulting oscillations.

Hope this helps move the investigation forward.

brunotl, 2025-10-16:
Thanks a lot for the detailed tests — that really helps.
It confirms what I was suspecting.
Looks like PirkerAnalysis must be refactored to properly handle "multi-leg" final glide.

Kamaltdinov, 2025-10-29:
I managed to fix all detected bugs and achieve correct operation in all of my test scenarios. However, I had to roll back the version changes made after 7.4.39 to the 7.40 AutoMC.cpp file, as it doesn't work correctly, even with multi-leg pirker.
https://github.com/LK8000/LK8000/pull/1664

closed 2025-11-26 by brunotl